sarnau/Inside-The-Loxone-Miniserver #3: Emulate Tree Device on real tree extension

challo2018, 2019-07-23:
Hello Markus, 
sorry for asking again, but i'm still trying to get your code running.
Is it possible to emulate an single tree device (eg. tree motion sensor) and connect SM32 CAN to tree bus to get the emulated sensor running with real tree extension. In your example, the emulated tree device is running together with emulated tree extension.

Thanks, BR Fabian

sarnau, 2019-07-23:
Yes, you can do a Tree Device only, just emulate that one without Tree Extension.

challo2018, 2019-07-23:
Thanks for your quick reply.
So simply:
devices.append(LoxBusTreeAlarmSiren(canbus, 0xbb008012))

sarnau, 2019-07-23:
I think that will do. Just make sure to not add any other LoxLink devices, that will mess up the CAN bus behind the Tree Extension.

challo2018, 2019-07-23:
good point! I'll check if it's working.
Thanks!!

closed 2019-08-27 by sarnau

chrisrock1984, 2020-12-22:
> Thanks for your quick reply.
> So simply:
> devices.append(LoxBusTreeAlarmSiren(canbus, 0xbb008012))

Hi,

I also tried Tree only device using -> static LoxBusTreeRgbwDimmer gLoxBusTreeRgbwDimmer(gLoxCANDriver, 0xb010035c, gResetReason);

but the device is not recognized by the miniserver