sarnau/Inside-The-Loxone-Miniserver #9: Loxone NAT - CAN Id

janbalcarik, 2020-01-06:
Good day,

thanks for the perfect study material.  
I've studied loxone NAT, but I still can't find out what device the packet is for.

Could you please give me little advice?

Recording communication
The packet is sent when the window opens: miniserver status

**Loxone Monitor**

00000042 192.168.1.77    15:38:55.609 15:40:02.506 LNK send poll packet to 14111118
00000043 192.168.1.77    15:38:55.609 15:40:02.506 LNK send standard packet: extension NAT: 5, device NAT: 0, Cmd: 0x05, sCommmandArg: 0x00, nData: 0x00000000
00000044 192.168.1.77    15:38:55.810 15:40:02.707 LNK send poll packet to 14111118
00000045 192.168.1.77    15:38:55.810 15:40:02.707 LNK send standard packet: extension NAT: 5, device NAT: 0, Cmd: 0x05, sCommmandArg: 0x00, nData: 0x00000000
00000046 192.168.1.77    15:38:56.010 15:40:02.908 LNK send poll packet to 14111118
00000047 192.168.1.77    15:38:56.011 15:40:02.908 LNK send standard packet: extension NAT: 5, device NAT: 0, Cmd: 0x05, sCommmandArg: 0x00, nData: 0x00000000

**My solutions - arduino + MCP2515**

RxID: 2422235141, RxID HEX: 90606005, Len: 8  Data: 00 00 00 00 00 00 00 00 
RxID: 2422235141, RxID HEX: 90606005, Len: 8  Data: 00 00 00 00 00 00 00 00
RxID: 2422235141, RxID HEX: 90606005, Len: 8  Data: 00 00 00 00 00 00 00 00

**My device serial**
0x14111118

I cannot find out by RxId for which extension the message is sent. In arduino I create 5 DI extension. 

Arduino is connected to 100 Papouch Quido inputs and processes the spinel protocol.
https://papouch.com/quido-eth-100-3-100-vstupu-3-vystupy-a-teplomer-p4641/?cid=145

Thank you very much for any advice.

John

chrisrock1984, 2020-01-06:
Hi

Can you please share your Arduino Code?

BR

janbalcarik, 2020-01-06:
I'm using the MCP_CAN library
https://github.com/coryjfowler/MCP_CAN_lib

```
#include <mcp_can.h>
#include <SD.h>
#include <SPI.h>

#define CAN_SELECT 9

// Define CAN
MCP_CAN CAN0(CAN_SELECT);
long unsigned int rxId;
unsigned char len = 0;
unsigned char rxBuf[8];

void setup() {

  // Inicializace Seriové komunikace
  Serial.begin(9600);

  // Inicializace SPI
  SPI.begin();

  // Inicializace_CAN
  can_Initialize();
}

long last_can_extension_Timer = 0;
void can_extension_Timer()
{
  if (millis() - last_can_extension_Timer >= 15000) // Každých 100 ms
  {
    // Nastavíme čas posledního spuštění
    last_can_extension_Timer = millis();

    // Spustíme dotaz
    can_send_online_all();
  }
}

void can_send_online(long quidoId)
{
  // send version extension 16.1.1.1
  byte aliveData[8] = {0x8f, 0x00, 0x00, 0x00, 0x75, 0x4b, 0xf4, 0x00};

  Serial.print("Device alive: ");
  Serial.print(quidoId, HEX);  
  
  byte sndStat = CAN0.sendMsgBuf(quidoId, 1, 8, aliveData);

  Serial.print(", Send status: ");
  Serial.print(sndStat);
  Serial.println();
}

void can_ReadData()
{
  //if (!digitalRead(CAN0_INT))                 // If CAN0_INT pin is low, read receive buffer
  //{

  if (CAN_MSGAVAIL == CAN0.checkReceive())           // check if data coming
  {
    CAN0.readMsgBuf(&rxId,  &len, rxBuf);      // Read data: len = data length, buf = data byte(s)

    Serial.print("RxID: ");
    Serial.print(rxId);
    Serial.print(", RxID HEX: ");
    Serial.print(rxId, HEX);
    Serial.print(", Len: ");
    Serial.print(len);

    Serial.print("  Data: ");
    for (int i = 0; i < len; i++)            // Tisk každého bytu dat
    {
      if (rxBuf[i] < 0x10)                   // Jestli byte dat míň než 0x10, přidat nulu před
      {
        Serial.print("0");
      }
      Serial.print(rxBuf[i], HEX);
      Serial.print(" ");
    }
    Serial.println();

    // Extension selection in Loxone Config
    if (rxBuf[0] == 0x08 && (rxId >= 2484146449 && rxId <= 2484146453))
    {
      can_send_online(rxId);
    }

    // !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

    Here I need to find out that the packet is for my device. See previous query.

    // !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
  }
}

void can_Initialize()
{
  delay(1000);
  //init CAN0 bus, baudrate: 125k@8MHz
  if (!CAN0.begin(MCP_ANY, CAN_125KBPS, MCP_8MHZ) == CAN_OK)
  {
    can_Initialize();
  }

  CAN0.setMode(MCP_NORMAL);  
  Serial.println("CAN complet");
}

void loop() {  
  can_ReadData();
  can_extension_Timer();  
}

```

sarnau, 2020-01-06:
You can't use a hardcoded ID for identification. There is a handshake protocol for the NAT between the Miniserver and the extension. This is documented in the Python sample code.

janbalcarik, 2020-01-06:
For testing, I have to buy USBTin. I guess I'm missing something ...
According to the Loxone log, I think the message is for DI extension.

I thought the log in loxone says it:
```
00000046 192.168.1.77 15:38:56.010 15:40:02.908 LNK send poll packet to 14111118
00000047 192.168.1.77 15:38:56.011 15:40:02.908 LNK send standard packet: extension NAT: 5, device NAT: 0, Cmd: 0x05, sCommmandArg: 0x00, nData: 0x00000000
```

I modified your code so that the receive method populates the address and data manually. The message does not reach the DI extension.

The console listing shows this message:

`<NAT    S6:E06 D00 05 b'00000000000000' Ping>`

janbalcarik, 2020-01-06:
Wow, I already got it ...

Thanks..

closed 2020-01-06 by janbalcarik

sarnau, 2020-01-07:
Cool