sarnau/Inside-The-Loxone-Miniserver #11: analyzeLoxoneExtensionUpdates.py not working

matlab22, 2020-02-01:
I used unpackLoxoneMiniserverFirmwareUpdate.py on 10031127_F2DE2A42CBBE36993AAF9BB8C2582498.upd to extract the data and got a folder named update_10031127 with all the extension updates like 10031127_1WireExtension_V2.upd.

After that I tried analyzeLoxoneExtensionUpdates.py to unpack the extensions. But  there is no 10031111_Miniserver.upd file. The only one I can find is a 10031127_Miniserver.bin but in a diffrent folder.
So i altered the code like this:
`
os.chdir("./update_10031127/")
analyzeUpdatefile('10031127_Miniserver.bin')
#analyzeUpdatefile('./update_10031127/10031127_Miniserver.bin')
#analyzeUpdatefile('10031111_Miniserver.upd')
os.chdir("./update")
for filename in sorted(glob.glob("*.upd")):
  analyzeUpdatefile(filename)
`
But then I am getting the following error:
`
analyzeLoxoneExtensionUpdates.py", line 230, in productByFilename
    encryptedBlock = binascii.unhexlify(product)
TypeError: Odd-length string
`
Is this script working for someone on the latest miniserver version?

matlab22, 2022-07-03:
Did anybody get this running?

matlab22, 2022-07-17:
@sarnau were can I find the requested file "10031111_Miniserver.upd"?

KjellVerb, 2022-10-17:
Same problem here...

KjellVerb, 2022-10-18:
I also commented the line `analyzeUpdatefile('10031111_Miniserver.upd')`, probably some folder structures changed in newer Loxone Config versions.

matlab22, 2023-03-05:
I added a try and except around analyzeUpdatefile(filename)
```
for filename in sorted(glob.glob("*.upd")):
  try:
    analyzeUpdatefile(filename)
  except Exception as e: print(e) 
```
I am getting a lot of 'float' object cannot be interpreted as an integer

There are a lot of unknown device. How do I find out what "product:"treeDev[0x0019]"" is?

```
13011117_0086CBCC2B89247CD354CBCDB1E06E56.upd                                   CPU:STM32F3        ROM:0x08001000-0x0801ab54  RAM:0x20000000-0x20007008 Category:NAT    Version:12.4.3.9   product:"treeDev[0x0020]" product name:"0086CBCC2B89247CD354CBCDB1E06E56"
13011117_00DA06A808F97215D28E60B9DD640B4A.upd                                   CPU:STM32F3        ROM:0x08000800-0x0800cf5c  RAM:0x20000000-0x20002c20 Category:Tree   Version:10.3.11.11 product:"Leaf Tree" product name:"00DA06A808F97215D28E60B9DD640B4A"
13011117_0158A919CF9CAFC61E376C846A0FAAF9.upd                                   CPU:STM32F3        ROM:0x08000800-0x0800b5fc  RAM:0x20000000-0x20006d38 Category:NAT    Version:10.5.2.17  product:"KNX Extension" product name:"0158A919CF9CAFC61E376C846A0FAAF9"
13011117_037E3AF3944DF6BC8596340791AC2C69.upd                                   CPU:STM32F3        ROM:0x08000800-0x080113e8  RAM:0x20000000-0x20004568 Category:Tree   Version:12.2.9.10  product:"Valve Tree" product name:"037E3AF3944DF6BC8596340791AC2C69"
13011117_06570C7AB317D8FF1FF138C3929C95A7.upd                                   CPU:STM32F3        ROM:0x08001000-0x0801fa38  RAM:0x20000000-0x2000a068 Category:NAT    Version:13.1.9.6   product:"treeDev[0x0019]" product name:"06570C7AB317D8FF1FF138C3929C95A7"
'float' object cannot be interpreted as an integer
13011117_0A57F5ABB36C95F12B5151538BD10F7B.upd                                   CPU:STM32F3        ROM:0x08001000-0x0801c7d0  RAM:0x20000000-0x2000c3b0 Category:NAT    Version:13.1.10.6  product:"treeDev[0x0041]" product name:"0A57F5ABB36C95F12B5151538BD10F7B"
'float' object cannot be interpreted as an integer
13011117_10FBEB6C19D8D27671BDE62F50FF2857110DDCA0FFBCF9CD3002674289CDA853.upd   CPU:STM32F3        ROM:0x08000800-0x0800d2a4  RAM:0x20000000-0x200045e8 Category:Tree   Version:13.1.7.21  product:"Nano Relay 2 Tree" product name:"10FBEB6C19D8D27671BDE62F50FF2857110DDCA0FFBCF9CD3002674289CDA853"
13011117_12FE58FAF29DF45D769032C5F334F302.upd                                   CPU:STM32F3        ROM:0x08000800-0x0800ebf4  RAM:0x20000000-0x20008e48 Category:NAT    Version:11.2.11.10 product:"Tree Extension" product name:"12FE58FAF29DF45D769032C5F334F302"
13011117_15619A11ADAD8F46D869FD820C6D05073EACE5FA33FCA17C75B9347FDFA7CF12.upd   CPU:STM32F3        ROM:0x08000800-0x0800ab28  RAM:0x20000000-0x20002680 Category:NAT    Version:10.5.1.15  product:"AO Extension" product name:"15619A11ADAD8F46D869FD820C6D05073EACE5FA33FCA17C75B9347FDFA7CF12"
13011117_16E05C3857F26D7D1451B5110309BDEF.upd                                   CPU:STM32F3        ROM:0x08000800-0x0800e538  RAM:0x20000000-0x20002b38 Category:Tree   Version:12.4.4.1   product:"Roomsensor Tree" product name:"16E05C3857F26D7D1451B5110309BDEF"
13011117_1A45A122D0EEF5A83657D5A141DDA0A2.upd                                   CPU:STM32F3        ROM:0x08000800-0x0800be04  RAM:0x20000000-0x20002a98 Category:Tree   Version:10.3.11.14 product:"Alarmsiren Tree" product name:"1A45A122D0EEF5A83657D5A141DDA0A2"
13011117_1A81AD2787299C25E9F81FE4C0D963B1115BD69E96881D96B522980A415BC208.upd   CPU:STM32F3        ROM:0x08000800-0x0801d1c8  RAM:0x20000000-0x200030b0 Category:Tree   Version:12.2.11.22 product:"treeDev[0x801E]" product name:"1A81AD2787299C25E9F81FE4C0D963B1115BD69E96881D96B522980A415BC208"
13011117_1A81AD2787299C25E9F81FE4C0D963B1CF9237963FB5F3D7EA86848E3028ED1B.upd   CPU:STM32F3        ROM:0x08001000-0x0801cff4  RAM:0x20000000-0x20008328 Category:NAT    Version:12.1.6.10  product:"treeDev[0x0045]" product name:"1A81AD2787299C25E9F81FE4C0D963B1CF9237963FB5F3D7EA86848E3028ED1B"
'float' object cannot be interpreted as an integer
13011117_1D8D8A40A26FC59B250F81A4F203BCE8.upd                                   CPU:STM32F3        ROM:0x08001000-0x0800b0b0  RAM:0x20000000-0x20003bf4 Category:NAT    Version:13.0.7.7   product name:"1D8D8A40A26FC59B250F81A4F203BCE8"
'float' object cannot be interpreted as an integer
'float' object cannot be interpreted as an integer
13011117_28781DC5F9FAE6F18B5267640A6279924FDB7D37D4F68F0B4155E343204619DB.upd   CPU:STM32F3        ROM:0x08000800-0x080100d8  RAM:0x20000000-0x20002bd8 Category:Tree   Version:12.3.12.17 product:"Corridor Light Tree" product name:"28781DC5F9FAE6F18B5267640A6279924FDB7D37D4F68F0B4155E343204619DB"
13011117_28D7AD3DE15857A4B5E36AC4B265A498ECEB0354F850614AF47A17242F988EFA.upd   CPU:STM32F3        ROM:0x08000800-0x0800b4f4  RAM:0x20000000-0x200026a0 Category:Tree   Version:10.3.11.11 product:"Downlight Spot RGBW Tree" product name:"28D7AD3DE15857A4B5E36AC4B265A498ECEB0354F850614AF47A17242F988EFA"
'float' object cannot be interpreted as an integer
13011117_2B38376F849190A14C7FE28F893C51B2.upd                                   CPU:STM32F3        ROM:0x08000800-0x080131bc  RAM:0x20000000-0x200037f0 Category:Tree   Version:12.4.3.10  product:"Touch Tree" product name:"2B38376F849190A14C7FE28F893C51B2"
13011117_2D2863FAAAD77FF21D63E3975227D11E1F8DFF153F4BFDDDEA8E8CE31454EE0B.upd   CPU:STM32F3        ROM:0x08001000-0x0801e290  RAM:0x20000000-0x2000c3b0 Category:NAT    Version:13.1.10.18 product:"treeDev[0x003D]" product name:"2D2863FAAAD77FF2 [... cut ...]
13011117_F84320193E8110282D09D62FCE754D9A9EC912FC3AE9DD351F9902ECF2FE4A7B.upd   CPU:STM32F3        ROM:0x08000800-0x0800db90  RAM:0x20000000-0x20002b00 Category:NAT    Version:12.3.12.17 product:"treeDev[0x0019]" product name:"F84320193E8110282D09D62FCE754D9A9EC912FC3AE9DD351F9902ECF2FE4A7B"
13011117_F84320193E8110282D09D62FCE754D9AABE6E4EC15984A823320985C3AE4C2BB.upd   CPU:STM32F3        ROM:0x08000800-0x08009b20  RAM:0x20000000-0x20002648 Category:NAT    Version:10.5.1.15  product:"AI Extension" product name:"F84320193E8110282D09D62FCE754D9AABE6E4EC15984A823320985C3AE4C2BB"
'float' object cannot be interpreted as an integer
```

matlab22, 2023-03-07:
I am a step closer, but there is still some numbers I could not figure out:
[image]
For example: According to the update list there should be a 0x0033
`Version:13.1.10.6  product:"treeDev[0x0033]" product name:"60EC75FA33507CF271445B01D4455763B952D01EB7562DEB8DD7C2E4BFADF399"`
but loxone list it as unknown