sarnau/Inside-The-Loxone-Miniserver #12: downloadLoxoneAESKeys.py trouble

feurer98, 2020-02-04:
Hey @sarnau nice work you're doing here.

I am currently playing with your code, but still without a concrete goal, maybe it should become a RS232 station. Now I wanted to read the AES keys with your Python script downloadLoxoneAESKeys.py. Unfortunately I always get errors here.

I assume that these scripts are based on Python 2.7.xxx, because StringIO from 3 on is no longer used.
Unfortunately I have problems with the ftplib. And since I have never written anything in Python before, I am a bit overwhelmed here. 
I entered the version number manually. My server is currently running on 10031125. 

My Errors:
`Traceback (most recent call last):
  File "C:\Users\Benjamin\Documents\Loxone\Planung\Inside-The-Loxone-Miniserver-master\Code\downloadLoxoneAESKeys.py", line 63, in <module>
    ftp.retrbinary("RETR 10031125_VmVyc3VjaE5yMQ0K.upd", callback=handle_binary)

  File "D:\Programme_Games\Python 27\lib\ftplib.py", line 416, in retrbinary
    conn = self.transfercmd(cmd, rest)

  File "D:\Programme_Games\Python 27\lib\ftplib.py", line 378, in transfercmd
    return self.ntransfercmd(cmd, rest)[0]

  File "D:\Programme_Games\Python 27\lib\ftplib.py", line 341, in ntransfercmd
    resp = self.sendcmd(cmd)

  File "D:\Programme_Games\Python 27\lib\ftplib.py", line 251, in sendcmd
    return self.getresp()

  File "D:\Programme_Games\Python 27\lib\ftplib.py", line 226, in getresp
    raise error_perm, resp

error_perm: 550 Can't open 10031125_VmVyc3VjaE5yMQ0K.upd: No such file or directory 
`

sarnau, 2020-02-04:
Sorry, I don't know what is wrong with your Python setup, especially on Windows.

feurer98, 2020-02-04:
@sarnau It didn't let me go today and so I still found the mistake today.
I can do with the "Loxone Gen. 2." Maybe it's different with Gen. 1?

In the query with nlst the first results are '.' and '..'
`['.', '..', '10031125_8C', .....]`
So I only changed one line of code from `version = ftp.nlst()[0].split(' ')[-1].split('_')[0]` to
`version = ftp.nlst()[2].split(' ')[-1].split('_')[0]`

closed 2020-02-04 by feurer98