sarnau/Inside-The-Loxone-Miniserver #14: LoxoneWeather.py as deamon

RubenWaterman, 2020-04-02:
hey! this LoxoneWeather.py is awesome and it works like a charm! How have you set this up on your raspberry pi so that it restarts if it crashes or that the python webserver restarts when the raspberry pi reboots?

sarnau, 2020-04-02:
I use http://supervisord.org

closed 2020-04-02 by sarnau

mabunixda, 2020-04-15:
I also created a [docker image](https://hub.docker.com/r/mabunixda/loxone-weatherservice) based one [this definition](https://github.com/mabunixda/dockerfiles/tree/master/loxone-weatherservice/)...

RubenWaterman, 2020-04-15:
@mabunixda That's super nice! Deployed it right away with a docker-compose file so that it comes back up after restarting as well :)

```
version: '3'
services:
  weather:
    build: .
    ports:
      - "6066:6066"
    restart: always
    environment:
      DARKSKY_SECRET_KEY: darkskykey
```