sarnau/Inside-The-Loxone-Miniserver #16: Loxone Nano DI Tree protocol

KjellVerb, 2020-04-09:
Amazing write-up, thanks so much for all the effort!

Do you happen to have documentation about the nano DI tree communication?

sarnau, 2020-04-09:
Product ID 0x800B, sends digital and frequency values, just like the DI Extensions. Limited to 6 bits – obviously. What else is there to know?

KjellVerb, 2020-04-09:
Good point, thanks. I didn't see a separate entry but all of the info is indeed there.

closed 2020-04-09 by KjellVerb

sarnau, 2020-04-09:
If not, let me know.

KjellVerb, 2020-04-09:
Thanks!
I'm looking into building my own extension clones and your write-up is perfect but I don't think I'm at the right level of programming. I'll probably use genuine Loxone components for the crucial stuff and diy extensions with UTP messaging for the bells and whistles.