sarnau/Inside-The-Loxone-Miniserver #17: Loxone Weather

mabunixda, 2020-04-15:
Because darksky will close its doors for api users, do you already planned to migrate the python script? 

I started to migrate your script to [openweathermap](https://github.com/mabunixda/Inside-The-Loxone-Miniserver/blob/d8d60ec85a18c564b7d48a5847ec48f5e224cd3e/Code/LoxoneWeather.py). But there are not always all values available ( e.g. rain ) Is this a problem for the miniserver when requesting the data?
Within csv I am reporting 0 on those values, on the XML I removed the tags.

sarnau, 2020-04-15:
Yes, I've used DarkSky because it specifically returns hourly forecasts for all needed values of the Miniserver. Almost no other free service offers that. A good solution could be [https://www.yr.no](url) - very simple xml data with one request.

Sungray, 2020-05-03:
https://openweathermap.org/darksky-openweather

I think openweathermap do return hourly forecasts, the switch should be pretty easy with this documentation.

sarnau, 2020-05-03:
Most of the documented values are completely ignored by the Miniserver. The forecast is only used for display in the UI. The current useful values are available in Loxone Config.

For anything relevant (rain, storm) you need a sensor on the roof anyway, because e.g. my report comes from Hamburg Airport (closes weather station) and we are a few km north of it. Neither rain nor wind is in any way accurately predicted here in North Germany. If, you you want to close windows or retract shades, you should really have sensors on your house.

Sungray, 2020-05-03:
Yes, why this function requires a subscription is beyond me.

sarnau, 2020-05-03:
Well, Meteoblue is quite expensive for individuals, I can only assume how much it cost for a company who wants 14 day forecasts for thousands of locations. But yes, for a simply weather forecast it is simply not worth it for me. Nice as a free feature, but actual data I look at WarnWetter with Radar, etc.

That leaves easy access to temperature, etc. – and I think you can finally feed these data in without subscription.

That leaves one thing: this annoying advertisements all over the consumer UI:
- the weather logo is always in the UI – toggling between different things, pulling the attention in (it is the only frequently changing element in the UI) – tapping it shows an ad
- the news feed is pushed into the UI
- now the "Do you like the app?"

Seriously? The Loxone frontend is like an appliance: a tool to do the job and should stay out of the way when it comes to the rest. Some stuff you can suppress, but it should be advertised in the configuration, not in the final UI! NEVER!

Woersty, 2020-05-03:
I use this one - works well.
https://www.loxwiki.eu/x/owASAg

markdegrootnl, 2024-02-22:
I'm using: https://github.com/jimmycleuren/loxone-weather-service