sarnau/Inside-The-Loxone-Miniserver #18: Problems using LoxoneMonitorServer.py

Woersty, 2020-04-18:
Hi, 
can you help me to get the Monitor Server Running?

```
loxberry@loxberrydev:~ $ python LoxoneMonitorServer.py
  File "LoxoneMonitorServer.py", line 31
    print('CPU  stats: Usage %3.1f%%' % (100-taskDeltaTime*100.0/cycleDeltaTime), end=' ')
                                                                                     ^
SyntaxError: invalid syntax
```
BR 
Wörsty

sarnau, 2020-04-18:
That is Python 3 syntax. Try `python3 LoxoneMonitorServer.py`

Woersty, 2020-04-18:
🤦

Woersty, 2020-04-18:
So ganz raff ich es noch nicht. 😭 
Haste Lust kurz zu telefonieren?
030 50158484

Woersty, 2020-04-18:
Angenommen ich starte das auf meinem LoxBerry mit der .50
und der Miniserver hat die .200
Wie startet ich das und was gebe ich bei LOCAL_IP an?

sarnau, 2020-04-18:
Es ist deutlich komplizierter: der Monitor Server (.200) muß laufen und der DNS Server für den Miniserver muß `monitorserver.loxone.com` auf den Monitor Server umlenken.

Woersty, 2020-04-18:
Okay. Umleitung von monitorserver.loxone.com auf .200 bekomme ich hin.
Aber wo wie was ist der Monitor Server?
Die Loxone Config? Ein Programm?
Ist das irgendwo erklärt?
Ich habe bisher nur das Script auf dem LoxBerry.

Woersty, 2020-04-18:
Die Loxone Config und der Monitor sind übrigens auf der .10 (Win 10)

Woersty, 2020-04-18:
> der Monitor Server (.200) muss laufen
Aber .200 ist der Miniserver
Und .50 der Linux Server mit dem .py
Und .10 Windows mit Loxone Config 
Und .254 mein DNS mit der Umleitung
Auf was muss ich jetzt umleiten?

Ich glaube das Probem ist der Webserver auf der .50
HTTPServer((LOCAL_IP, 80), Proxy)
Ich schalt den mal ab - moment

Woersty, 2020-04-18:
Jetzt läuft der Server aber ich sehe nur http Anrfragen, die ich an Port 80 schicke auf die .50.

Woersty, 2020-04-18:
[image]

Wie es aussieht muss Python dann aber auf dem gleichen Server laufen wie die Loxone Config?

Woersty, 2020-04-18:
Lööft. 
```

root@loxberrydev:/opt/loxberry# python3 LoxoneMonitorServer.py
Starting Monitor Server Proxy...
Starting Monitor Server Logger...
2020-04-18 11:37:02.408 %s ['loxberry.woersty.com', '192.168.178.254', '192.168.178.254']
2020-04-18 11:37:04.984 Stream connect error 192.168.178.50:80
2020-04-18 11:37:05.012 %s ['loxberry.woersty.com', '192.168.178.254', '192.168.178.254']
2020-04-18 11:37:27.141 %s ['loxberrydev.woersty.com', '192.168.178.50', '192.168.178.254']
2020-04-18 11:37:27.451 ARP of 192.168.178.50 : Time 312428us - failed
2020-04-18 11:37:27.451 Stream connect error 192.168.178.50:80
2020-04-18 11:37:28.943 %s ['wetterstation.woersty.com', '192.168.178.206', '192.168.178.254']
2020-04-18 11:37:28.983 HTC no length specified
2020-04-18 11:37:29.059 TCP ceb2a8c0/80/13339 sequ error rcv=2215630 exp=2217056
2020-04-18 11:37:29.495 TCP ceb2a8c0/80/13339 sequ error rcv=2226268 exp=2227444
2020-04-18 11:37:31.275 HTTP Client HTTP5 connected, Socket 77 Port 40306
2020-04-18 11:37:31.389 HTTP Client HTTP7 connected, Socket 102 Port 40308
2020-04-18 11:37:31.403 HTTP Client HTTP2 connected, Socket 212 Port 40310
2020-04-18 11:37:31.421 HTTP Client HTTP3 connected, Socket 239 Port 40312
2020-04-18 11:37:31.435 HTTP Client HTTP6 connected, Socket 97 Port 40314
2020-04-18 11:37:31.451 %s ['loxberrydev.woersty.com', '192.168.178.50', '192.168.178.254
```

Woersty, 2020-04-18:
Wenn wir jetzt noch einen Weg finden, das 
[image]
irgendwie selbst zu triggern - und zwar auf die richtige IP wäre das cool.
Denn ich musste jetzt 
-Den Loxberry stoppen
-Meinem Windows PC die IP vom LoxBerry geben
-Den Monitor starten + nach Verbindungsende
-Loxone Monitor und Config schließen
-Meinem Windows PC wieder seine IP geben
-Den Loxberry starten
-Apache stoppen
-python3 LoxoneMonitorServer.py starten
Bissel overkill und geht max 7 Tage ...

Woersty, 2020-04-18:
Was ich nicht verstehe ist, sollte das nicht eigentlich über monitorserver.loxone.com gehen?
Also den externen zu mir?

Woersty, 2020-04-18:
Einfach folgendes senden `http://miniserver/dev/sps/log/192.168.178.50`
Dann gehts

Woersty, 2020-04-18:
Also ich fasse zusammen:
-Miniserver läuft auf port pp an IP mmm.mmm.mmm.mmm
-Python Script läuft Port bb auf Server aaa.aaa.aaa.aaa:bb 
Man ruft auf dem Miniserver folgende Adresse auf, um den Monitor umzulenken: 
`http://mmm.mmm.mmm.mmm:pp/dev/sps/log/aaa.aaa.aaa.aaa:bb
`-Man editiert das Python Script und trägt als 
LOCAL_IP aaa.aaa.aaa.aaa ein und bei Zeile 220 den Port bb statt 80.
 `httpd = HTTPServer((LOCAL_IP, 81), Proxy)
`Dann kann ein anderer Webserver weiter auf :80 laufen - wie am LoxBerry z.B.
-Dann Python Script aufrufen:
` python3 LoxoneMonitorServer.py`
Vielleicht kannst du dein LoxoneMonitorServer.py ja einfach so umbauen,
dass man die IP und den Port als CLI Option mitgibt.
` python3 LoxoneMonitorServer.py ` hostname -I` 81`

flibbi2, 2021-04-14:
In der aktuellen Version kracht das ganze leider..... liegts am neu hinzugekommenen Audioserver?

2021-04-14 15:09:51.360 MSE data received: 82 Block:0 Len:160
2021-04-14 15:09:51.361 MSE received:160 from 10.199.199.7/7076
2021-04-14 15:09:51.361 MSE from slave 16e9eac3-03a5-7dd1-ffff9a8dad1579a8 1.000 0 00 00 00 00
2021-04-14 15:09:51.362 MSE from slave 16e9eac3-03a5-7db8-ffff8085bc30bae1 1.128 0 00 00 00 00
2021-04-14 15:09:51.362 MSE from slave 16e9eac3-03a5-7dc8-ffff8085bc30bae1 -8.876 0 00 00 00 00
2021-04-14 15:09:51.363 MSE from slave 15f22a5f-0008-c262-fffffd62f892a111 1.128 0 00 00 00 00
2021-04-14 15:09:51.364 MSE from slave 16a6e1ea-0268-9a73-ffff7df86fff6232 1.120 0 00 00 00 00
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "LoxoneMonitorServer.py", line 195, in monitorserver_logger
    data = parsePackage(data)
  File "LoxoneMonitorServer.py", line 180, in parsePackage
    parsePackageContent(pkg)
  File "LoxoneMonitorServer.py", line 162, in parsePackageContent
    parsePackageLogging(pkg,strType,parameterCount)
  File "LoxoneMonitorServer.py", line 108, in parsePackageLogging
    while pkg[offset]:
IndexError: index out of range

^C2021-04-14 15:09:51.394 Traceback (most recent call last):
  File "LoxoneMonitorServer.py", line 231, in <module>
    x.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):

Gibts dafür eine Lösung?