sarnau/Inside-The-Loxone-Miniserver #20: Simulation eines Miniservers

klenzel, 2020-06-12:
Hallo,

vielen Dank für deine Mühen, uns tiefere Einblicke in die Hardware von Loxone zu gewähren.
Nach dem Lesen deiner Beiträge frage ich mich, ob es nicht möglich wäre, zu Testzwecken einen Miniserver zu simulieren. Dieser hätte zwar zunächst keinerlei Ein- und Ausgänge, könnte aber ein Projekt ausführen.

Wenn man beispielsweise ein Modul wie "Rock64" nimmt, hat man die CPU des Miniservers Gen.2 sowie einen SD-Kartenslot und Ethernet. Du hast einen Flash-Speicher beschrieben, der eine Art Bootloader beinhaltet, der den Inhalt der SD-Karte ausliest und in den RAM legt.
Meinst du, den Flash könnte man auslesen und somit einen Miniserver zum Testen emulieren?

Grüße
Daniel

sarnau, 2020-06-12:
Könnte man – aber mal ehrlich: warum? Die Hardware nachbauen und Firmware kopieren – das kannste nicht machen, denn das wäre eine Raubkopie. Zudem steht der Aufwand in keinem Verhältnis zum Ergebnis. Kaufe einen gebrauchten Loxone Go und fertig.

Wenn Du bestimmte Loxone Extensions nutzen bzw. testen willst, brauchst Du nur den CAN Bus simulieren. Das mache ich mit einem Python Simulator. Mein Ziel war es das Loxone System zu erweitern, nicht zu kopieren.

klenzel, 2020-06-14:
Guten Morgen,

ich wollte deiner Arbeit auch keine böse Absichten unterstellen. Mir ging es mit meiner Idee eher um eine Spielerei als darum, jemanden um seine Entwicklung zu bringen ;)

Grüße
Daniel

closed 2020-06-14 by klenzel