sarnau/Inside-The-Loxone-Miniserver #23: Loxone Air

chrisrock1984, 2020-11-05:
Hi, 
it's not an issue but more a question about Loxone Air.

Are there any plans to reverse engineer the Air technology as well?
Would be interesting, if the modules used by Loxone can also be used to create a DIY Air product ;-)

Thx

sarnau, 2020-11-05:
Loxone uses a custom chipset for this (I've documented it and you can even download the official SDK). The chipset is rather expensive (I think 20 Euro), which makes it quite unattractive. The SDK is pretty much taken as-is from Loxone and they added sometimes only minimal code for the feature. I do not think it makes a lot of sense to waste time here, it's not like you can replace it with a 2 Euro Blue-Pill.

chrisrock1984, 2020-11-05:
true!

closed 2020-11-05 by chrisrock1984