sarnau/Inside-The-Loxone-Miniserver #25: Cannot format SD card with Loxone Config: Error 12

cmonty14, 2020-12-07:
Hi,
I cannot format the SD card with Loxone Config; the error displayed is:
Fehler - 12 beim Formatieren

[image]

I have tried to format the SD card with several card readers w/o success.
Then I started to format the SD card with Windows tools, and this worked. Now the card is totally empty.

Can you advise how to fix this?

THX

sarnau, 2020-12-07:
That is more a question for a Loxone forum, like [https://www.loxforum.com](https://www.loxforum.com). Anti-Virus software or something?

closed 2020-12-07 by sarnau

cmonty14, 2020-12-07:
After formatting th SD card with Windows tools, the error message is now:
Fehler - 1 beim Formatieren

Can I still use the SD card although is is completely empty now?

sarnau, 2020-12-07:
No idea, but if the SD card is not defect, it will be usable. But I am not using Windows.

cmonty14, 2020-12-07:
I would prefer to use Linux OS, but then I must use Wine.
Do you know if I can format a SD card with Loxone Config in Wine?
Or can you recommend any other way to format the SD card?

Woersty, 2021-04-18:
I know it's a bit late but I had the same. You need to be Admin when starting Loxone config and you must try it several times sometimes.

-1 means wrong file system format
It must be FAT32 but not NTFS.