sarnau/Inside-The-Loxone-Miniserver #28: Minimal Loxone Setup

cody82, 2021-04-08:
I am currently searching for a wired smart home system for light switches and motion/presence detectors. As I already have a zigbee system with Node Red and Home Assistant, I only need the devices, not the server.
So is it possible with this software and a CAN adapter to use the Loxone devices? Or do I need the Tree Extension?

I also noticed the topic "Authorization of Extensions and Tree devices". Do I need this if I do not use a Miniserver?

sarnau, 2021-04-08:
Why do you want to do that? That makes very little sense – if you use KNX components, you have an open system without a central server. And it integrates into every home automation system.

cody82, 2021-04-09:
Right, I am also looking into KNX. But I would not call it "open". As far as I understand, you have to use the extremely expensive proprietary Windows-only software ETS to configure the devices. I found no alternative for it. Which is kind of strange that there is more DIY support for closed systems on the net than KNX. Is the KNX configuration process too complicated to implement even though the standard is kind of public? Or am I the only one who is interested in this?
So yeah, apart from the ETS problem KNX looks good.

sarnau, 2021-04-09:
ETS is an issue, but it is still cheaper than Loxone – especially if you buy more than just a few components. And all the communication between the devices is standardized, which is great.

jezikk82, 2021-04-09:
There is free ets version for up2 five devices in one project. On small projects (in my case 12 devs) you can create 3 projects (5+5+2). The only downside of that is you have keep track of the addresses outside, eg in excel and interconnections (between devices in other project) you make not by dragging but by putting the address manually.

christianTF, 2021-04-09:
"Extremely expensive" is relative.
Loxone Miniserver costs around ~750 Euros, every single device around 250 and 450 Euros.

ETS Free is free (5 devices each project), ETS Lite ~250 Euro (20 devices each project), ETS Home around 450 Euros (64 devices). 
You don't have all the proprietary limits (c) by Loxone, but 500 manufacturers and 20.000 devices to choose from.
Your ioBroker and Node-Red have built-in KNX support. As your logic is done on Node-Red, it is not very complex to configure (1.000 tutorials on the Web).

cody82, 2021-04-09:
Yes you are right. Although for Loxone my initial plan was not to use the Miniserver but only the Three Extension.
So I will go the KNX way. Thank you all for your help!

closed 2021-04-09 by cody82

sarnau, 2021-04-09:
BTW: for my wife a decision point was simply the light switches – she didn't like the ones from Loxone (and if they ever change the design: you had dead in the water, if you need replacements). With KNX we could choose (and it was even cheaper)