sarnau/Inside-The-Loxone-Miniserver #29: loadMiniserverConfigurationFile.py has problems with Logos

Woersty, 2021-05-14:
It seems the loadMiniserverConfigurationFile.py has problems with some Logos.

Example Logo: https://github.com/mschlenstedt/LoxBerry-Plugin-Stats4Lox-NG/issues/64#issuecomment-841115641

It results in invalid xml after the parsing like this and other strange stuff:
`Logo="89504E470D0A1A0A0000000D49484452000000D2000000780806000000871EACCA000000017352474200AECE1CE90000000467414D410000B18F0BFC6105000000097048597300000EC300000EC301C76FA86400001`51149444154785EED9D79705455BEC7032220041208100C0112202C41E0C13C1C75D4D1575A4F105CFE70667C3265E9F3B961890F9711049FE286C8328E58F3906795A5E5E894BBD6302AE30CE21BDFE01650645FB292A4FB7677964E4267E1F7CEF7E49EE4A673FAE676BA3BE99BFCBE55DF4AD2B9F7DC73EF399FB3DDD3E7A4503F54D38F07EDB3F33E6C56A7726DF19233313FEFFEB1F1FBEA055217E7B1813FEACC3DE9547B5Er p42F4A0A00000DFE82BCE963C9373E9B0273E69B2099167F1B99D9E4CBCEA5C61FF69B67B058D1A9CF8354BFF9B7E4199E46FE693329709EA885F2E77604097F8BCFFDB9D3C948CBA0FA6DDBCD33592CE7EAD3B9F7DCF2EFE44D19487E00A4A0B142A46CF9BCF2ECA154B36E9D19028BE54C7D12A496DA2019E78AE6DAB8ACCE4DB9AE8CA65E7A06F966E4538BDF6F122062D9ABCF81D4F0F63BE41D355636D564534E078B9DDB9A7A79E43927954EFFF9CF66C8BE55D4F529906AEE5D214E58FC3366EB217B3F33A9E7CF9F479E21A9145CF35FE615582CBDFA04482DB954749BFFCF64A48DE95E2D6467119E374D34F55DF4A696AA6AF38ABE5547B91EA4D05F7791317112F972A6B54214694021168B707D93A79231368B425FFCDDBC328BD52E578354F7FC0BE449398BFC33CF6BCDF0898020627F5E3E19A2E918DCB8C58C018BD52A5782D4525545FEF90BC918369202731768337DC23C67017905BCD5BF5C66C686C5722148A1FFFD3B19991344732EB775683BC1B55027E37A18221F9F4DC1DFAC3663C5EAEF721548C1D56BC9337818F9A7E749FEEAF08CDE5316D7F6A5659077FA4C3376ACFE2CD78014F8F9BF90913AAAFD052B0000B72593C7C76678BA70DB3E9B4715C38653E8CB2FCD98B2FAA3921EA4E61327657FC837613205E6FDA4B54FA4B36AE6C118BDD31D13ADE528A00524DD31B0B8A6317A1CD5FD6EAB196B56E676528354FBD000002225853CC25E1BB7FEFFEC56D044E637C64F949FC56A5FB680175FB71010C9990EE233753D9DCB85FD0B7F4A4DFB0F9877C0EA2F4A5A904ED7D551DD7BEF53C38103D470F060EBCF086E3C74587E55C2481F47FEC95329B8F6516A2A2A263A7182A8BC5CB0222A37EFDD4B4D478F52F5AF96913F6F969C395E7DC522AA2F2BA586FDFB89F6EDA3A682026AFCEEBB8EDEBB8F4296795A8D7BBE36EF82D55F949420B5B4B450E9A9535414F05391CFD7A50BAB02541F0890312455D61EDE71E36538FF76FBED943525977266CC70ECA16969F25CA8F6CEE4749F325DBA66F1B5D4203E2B320C3D4D1646A75B0F61B1A4920EA4C6C6462A2E2E962E1319B6ACB4B44B977A2AA941D424BEA123445F4934ED32CEA5A63FBE45350D0D94229A5BD1FAF1C71F9771A9BDFD6E13A419145CBC943CE2B31211AF5302F213A2B60B0878592C286940422D6488D2FEinfo43545858287DF2E4C988C6FF015AA902A9C20449F491FCD36753F0CA4574CAEBD582D2956FBAE92619271D48278E1F6FBBFEB163C7E4EF809FD57B3A2ED2E453D1A4DEB56B177DFEF9E7F2E78A152BE876D122B9E38E3B6C8D63EEBAEB2EAAAE8E6D1E65D28084D2BDBEBE5E66CA5028646B1C8363254C3A90449FA6FE97CB575A0053074A57BEF9E69B659CDA409A3A83AAAFBA9A9AC58E58716910B5DEE9D3DCD0EB4DAD59B3469B8E4E3D70E040F28A42371625ED6043576A6A6A92B54159A9A895B420DD145790AAFE7589FC8C957C7AE28927B4E9E8D49FB38EA4AAAA2A33B4EEC9D520B5D6480C527F17831483182496523C408A75E08841D2984172975E79E5156D3A3AF5D8B163CD90BA2F4720BDF3CE3BB47CF9727AECB1C7B41000B491B1376EDC48478F1E35634472940543D072E4AE0741C2488FEE1CA77EF5D5576538D0A64D9B283F3F5F7B5CB4CECCCC1104EEBB8FDE78E38D6E3753BEF9E61BBAEDB6DBE8DE7BEFD55E23D15EB66C99CC67E8FB46D2AC59B3B4E7F6862FB9E41233565D80F4E8A38F527A7ABA3690DEF24B2FBD2447ECCE9C3943E4D155F27D534F828421FAfo43876ACF73E2B3CE3A4B8673D1451769FF1F0F237EFBF6ED93D771A2AD5BB7D2F8F1E3B561F596EFBEFB60CE2EE14AE4738BD6D75D779D192B1B90962E5DAA3D3919BC70E1423396244BAF624D414F36ED1E7EF861ED794E7DE185176A3F8FB73FFAE82333C691F5EEBBEF6ACF4D1687EBD24B2FD51ED71BBEFEFAEBCD58450069CB962DDA1393C9A8E2950A45C9D59320BDFCF2CBDAF392D178D91849DBB76FD79E934C1E3D7A7487F7743D550839312A1BA53690D0548276ECD8A13D291A0F1A3488264F9E4CD9D9D941549C38312E4D8FB29F7F5FC63900005250F4957A0A24342F75E745EB8C8C0C9A346952C4673461C204ED79D11A4DB770EDDDBB577B6CB49E3A75AA36FECAB9B9B9DAF3A2F1534F3D65C69A68D1A245727040858F7C865137DD79D118CF1BB6C65D67A4090CC06FB9E5163356613512660C60AAFC61424E8DCC81D13427BAF3CE3BB5613835A67740A74521102C2923C3e" WDD78E38D6668F6FAE1871FB4E747E3D5AB3B7E25BEAEAE4E36FB74C73A3532EFDB6FBF6D8668AF5887A731C003459A41F2E28B2F6ACF736AE4D9582541C23C374CC43C72E488EC0CEB2EE6D4B82908236AE8BF4472494909D5D6D6C9705DA566D1B43386B80724BFDF2FEFDFEE39151515C9819578F463D4E003D219D21D138DAFBCF24A190E4652757157C9704428D69A494D620E7F5E7886CF3CF38CF61CA74E4B4B239FCF27C34325600DDF6ACCEB8C24991391A91149043262C408EDC59C7AEDDAB53260640239A1D462399BDB3412143F75614463D4D67277818467DDDCDCDC9640BA67046394F0B5D75ED386E1D428AC0E1E3C286B225C0FD21D178DD5D02FD211F927D23DE07F28A86369A60E1F3E9C2A0B1889E7D5F6AD00615C0B05CDB3CF3EAB3DCFA9EC300A36EB3D841BF751595929EF59A71454973elekF048EB5BD8921730819253C22284DF013C6677818BA30A2B1921B4142E65019024626C14C66F56C60949478F7A40BC3A901D2E1C3876544674107E98E8BC6183D835023A9F8C3485BC0AAD219D78B15A461C386C9B86362A90A37DE20A9B0ADF76235D206FF8FA414548D3848D5178082F4FCF3CF9B41EB8552050F1642098999B7BA709C5AC96D20A99A215C98B1612D75E30512BE9E826930F102E98A2BAE90E14412D25919C2A0802E1C270648C89F1006C5F07C54069EE448C8F34E5E62AB01399D5214D978F985871C6BD30E0F189DDBFBEFBFBF931F78E470DAE8A187E1F7AD5A45B292A40F0D1830401B8E532BF5159020948E0AA67883847021DD71D118A38D687DE8D2F9C1071F6C4B67185F948C255F5941424163AD39EA7722C2A4FD000F124CCD1288B5464A692BF5259082C1A02CD49219A49E744F8014F3EC6F74B2103106A9DDBD0D926A6E3348AD7605488814D7481DCD202597316AA7E6DD252D48482C06A9A319A4E4B37A1FD56F40C2AC5D0CB5EA77E0B29F7F1F777FFBEDB7B467CF1EFAEAABAFE422174A00526427071103CF3F5F0EE117141468D32956630A13D25919EB6340FD06A4A796795A068C9127F4BF9011E265BC9C8CB4620F8314D98900E9B2CB2E93E1A0EF12EF748631E0A253BF0149CD6C4058EA66E3658C64D51144C8F0C3C52045762240522F64711F88A72EBDBA6B801269559F44818467138BDA40C25B76B443637D41BA61C3061930DED0E321C7D308535772A0D2AF2F3BE5AA49ABB89F48720348975F7EB90C07F333D1C40B4FAB588C19301E0F5611EC2C1D48EBD7AFD7C6D1A9478D1A25C389456D20C1A84ED7AD5BA7BD9853DF70C30D66D09DD7808B87959078070EB42E561F100F21582A321D83A4B51524643ED4EACF3DF79CF658A79E313F76DBAC055D3AC582EB2C6F341D01AB523848981D136B8D3464C8103917B1BBFAE4934FDA4142A470032FBCF082F662D178E6CCC46DBE8537E78B172F96D7C19A0750514579FB92C500522787838466D3EBAFBFAE3D361AA34984053213A127967C527E2911D7C100969215243C1F80863860883C3C7ED1180583936F145B85991D3939397265DE0E20A1338FEF23C5DABCEB296374072AA9ACD48014879556EFE89B20A1D984854E74C7269B515B7CFCF1C7E6936905493D1B1850A102B8F5D65BB5E7F7843FF8E083F617B20A2634EFB06A90EE846432BE9909A1CFD4E9ABE679B3A673C9B5541E2348D5BFBE59420998AC20C59AB161F5A65E272B48C8F46FBDF596360CA71E3C78B000EF0525D218C6353EFCF043EDF1C9E4BCBCBC0E934531E1197956E55BDC0B6686A38F3366CC186D1889F69B6FBE49298844384C109676D29D942CC6244848CE4B0B03C977EE446ADCB4994E8A7BD19DDB95553FAF5E347F7C19599D6AA45897E482B15E42242990902ED092254BB4614463C08F4CA80055617FF6D967DAE393C5A9A9A99D16B85783262ACFE25ED057423F471746228D810ABC924901C96A82A4B26A76ECDEBD5B7E25BAB748B7B32AD131D2D861392EEC8F34F81CF9BFFFF9C31FE8C90D1B68F3D6ADB449F4FD9C78FDE6CDB463E74E793EE41D964AFEDCF61A091D5F5D7CA23596CC8AA43A51381C292CA432718FA84574E7476BD56F553512D21999119900A3642B5DAE8CDB1A11F1F44E4B5A288583045B5F8D6020E5820B2ED086176F631DE448CEB101D1EAE15A238692000F1A2FDCF6EFDF2F9B1958E76ECA94293D6E7C55191D3B7CAF051D4B55A22213947A154823C937791A051F5B47CD65A26645ED8A6154015A5416E7612A4F84117C7A3D1919E3A9EAAA5690B08062565696368E7646FC1177652C9EF1DE7BEFC930ADC237B530405D5F5020FF5EFDC823346EDC386D987656CF4A19E986D125149CD64C887446FAA2EF81DA1D23A19849306700D419862EEC445BC51DA5FDB66DDBE47308972C40C3F22CEE0BEF43D167427310F782CFD0824021A1BB56B4B6E643E4838B2FBE58C6478284A6046A216BA460645618A53F4A46BBAFDAF6B450FAA77EB21010CE21AD469ABB808C71595439680879CE8ECD95830693em="9A76A2CF552340C2E07B438CEF1BBA52D33FBEA2A080B5529476CD057B29F257C9BA27BC2F0C6F812043AAC254CD4849760116598886C184BF010F5E8FA0E0875148245AEDB33E85D44B5444441741F5B3378C78A11DAC847718127ECDDADFF136061BB0D1188A9162910975F1EBCA7876EABD4BB89A8E1CA5E0AAB5E41D318ABC2903E4e" L3162349D3974987C667F49176624030A94CC98100890D608D79AC6BDCE2ECAD6265A57C2F3C4BDAA250CC2EF25DEF763F75C3B8084CE28322B0EC60D29E3812390DE32E2103ECA855206A58E1624B59D7F9CAC76ECC3A49522714D511CBB3276FA0B5A0A8296AA2AAA7BF1F714F8D92564A48F11355F36F967CCA6C09CF9F21E8CD4746A119D676F4DB95C4F5D98918CF482F1723392D0C453D3C292259D717DDC6BB4421704D058EF2511F78330711DDD57CE3B806415AA4E3C6CFC4409D65B4662A3A9810EA6F5067A1AA4DA454B09939300B42E9EB616F107848670F3D7DF5060C9B5648C194F465A06F973F33AC7D904A9F1C7035429328936CC2E8C66389E5957421F18E98C979A68BEE379EBC2EB09A319863874572ACFE25EF7E45F779813F8C [... cut ...] 706C9A11824969D1824876290587662901C8A4162D9894172280689652706C9A11824969D1824870A07093BF661EB4B2CD21F0F07F26101A509A6312C8D417291182487EA0052453935EDDA4D819119E4CFCE89D90161DF84C96464999E90439E9401D4F4FD4FB7EAAC641783E450569081549D22FFFFEDA1C09EAFA9A6EC14559794C4E8526A292EA633274E52F3F113D24D070FD19950C8BC3A2BD9C52039941524AC2B5EECF5525165251589DFE3E163858554DF847D03596E1483E4501D4032619256E6C768AC636EDD6A93E52E31480ED509A4381A9B3C23EC9E5A7E99156F11FD3F0B20CE97E2F729D48445200049454E44AE426082" `

Woersty, 2021-05-14:
[image]

christianTF, 2021-10-05:
The issue seems to concern all Base64 binaries in the LoxPlan XML.
We also got an issue on unpacking with SVG icons in the XML.

sarnau, 2021-11-14:
I've fixed it, see my reading code for the project file

closed 2021-11-14 by sarnau