sarnau/Inside-The-Loxone-Miniserver #30: Adding iButtons to Users

alexanderkraemer, 2021-08-02:
Hi Markus,

we've been trying hard to figure out a way on how to add iButtons to a User through the loxone API.
Officially it's not supported, altough Keycodes and NFC Tags are.
We tried to reverse engineer their Miniserver API by sniffing to the network traffic without any success.
Do you have an idea how this can be made possible?

Thanks!
Best, Alex

alexanderkraemer, 2021-08-22:
Hi Markus, hast du diesbezüglich zufällig eine Idee wie man das Problem lösen kann?
glg. Alex

sarnau, 2021-11-14:
Habe ich mir nicht angesehen. Es ist möglich, das API unterstützt das natürlich – sonst könnte Loxone Config es ja auch nicht. Aber wozu willst Du das? iButtons sind inherent unsicher, weil sie quasi nur eine Seriennummer sind.

alexanderkraemer, 2021-11-14:
Nun, loxone config schreibt das ganze ja nur in das XML file rein, welches dann auf den Miniserver geladen wird. Dazu muss eigentlich nichts per API gemacht werden, oder?