sarnau/Inside-The-Loxone-Miniserver #32: Unpack of LoxCC corruption with large base64 content

christianTF, 2021-11-13:
Hi Markus,

I've re-created a new issue for the Unpack LoxCC corruption issue as the unpack from Inside-The-Loxone-Miniserver is a base component of the LoxBerry Plugin Stats4Lox, that directly parses the Loxone XML on the LoxBerry to show the elements the user can collect for statistics.  
We noticed problems with the unpack algo with users that have large base64 content in their LoxPlan file, like a company logo, or own SVG icons. In this case, the unpack generates a corrupt XML file.

I've reproduced the issue on my Test-Miniserver and have uploaded all the data I could fetch:
[s4l_loxplan_ms3.Loxone.unpacked_corrupt.zip](https://github.com/sarnau/Inside-The-Loxone-Miniserver/files/7532461/s4l_loxplan_ms3.Loxone.unpacked_corrupt.zip)
[s4l_loxplan_ms3_downloaded_from_ms.zip](https://github.com/sarnau/Inside-The-Loxone-Miniserver/files/7532462/s4l_loxplan_ms3_downloaded_from_ms.zip)

In this issue included:
- s4l_loxplan_ms3.Loxone.unpacked_corrupt.zip --> This is the corrupt resulting XML
- s4l_loxplan_ms3_downloaded_from_ms.zip --> This is the most current ZIP file from the Miniserver (original `/dev/fsget/prog/sps_0203_20211113182248.zip`)
- sps0.LoxCC_in_MS_ZIP.zip --> This is the LoxCC file (out of the zip above) that should be unpacked 
- Test-Miniserver.loxplan_from_PC.zip --> This is the orginal LoxPlan file copied from the PC (--> the "should be" result of the unpack)
- unpack_loxcc_py3.zip --> This is the unpack routine we use, an modified version of `loadMiniserverConfigurationFile.py` without file download, but only extracting a local LoxCC file

We and our users would be very pleased if you could figure out what the issue is with the LoxCC unpack routine.

Thanks a lot!
Christian

[sps0.LoxCC_in_MS_ZIP.zip](https://github.com/sarnau/Inside-The-Loxone-Miniserver/files/7532463/sps0.LoxCC_in_MS_ZIP.zip)
[Test-Miniserver.loxplan_from_PC.zip](https://github.com/sarnau/Inside-The-Loxone-Miniserver/files/7532464/Test-Miniserver.loxplan_from_PC.zip)
[unpack_loxcc_py3.zip](https://github.com/sarnau/Inside-The-Loxone-Miniserver/files/7532465/unpack_loxcc_py3.zip)

sarnau, 2021-11-14:
I fixed this issue – either it was an oversight or they changed it for a better compression ratio. Either way. I've also added additional checks for the CRC and generated file size.

closed 2021-11-14 by sarnau

christianTF, 2021-11-14:
Thanks Markus, tested and works perfect!! :relaxed:

sarnau, 2021-11-14:
Looks like it was an oversight – really weird, that I missed that. Anyway, with the CRC32 now being checked, the validity can now be guaranteed.

mschlenstedt, 2021-11-14:
Thanks a lot!