sarnau/Inside-The-Loxone-Miniserver #33: info about Cloud Service Caller still up to date?

dimmu311, 2021-11-24:
I'm currently working on my own implementation of the cloud caller. i redirect the dns queries from caller.loxone.com to my own web server.
To the request <http: //caller.loxone.com: 80/cgi-bin/loxlive/call.pl? ......> I answer according to the given example. In addition, information on the cloud caller says that the miniserver ignores the answer.

I think that is not correct (anymore). Immediately after the request is sent to the caller, an error message appears in the Loxone Monitor and a push message is issued that the caller is not active.
See this example from Loxone Monitor:
2021-11-23 22: 57: 22.658; Critical 509 QUITTED, MINISERVER_NAME - Error Loxone Caller Service, all, Miniserver (Uuid 1111111111-0000-2222-ffff141234567890)

I think the miniserver is expecting (maybe in newer firmware) an answer.

bate, 2021-11-24:
Thats easy, just proxy the request to the correct loxone server and reply back to the miniserver to track the response.
But thats an http call without ssl check .. ok easy then.

dimmu311, 2021-11-24:
I thought about that too. But there is a little problem.
The original server checks whether a caller license is activated for the Miniserver Mac address.
So either I need to know a mac address that has an active license or I need to buy a license