sarnau/Inside-The-Loxone-Miniserver #34: Does the Modbus extension use read/write multiple registers

raintonr, 2022-08-03:
The Modbus protocol caters for reading or writing of a contiguous range of registers or coils in a single command... do you know if Loxone actually uses this feature or always reads/writes a single register or coil multiple times, even when address ranges are contiguous and have same polling frequency?

Ie:

Say I have a sensor device with 2 inputs on holding registers 1 & 2 which are both configured to be polled every 10 seconds.

Making full use of the Modbus protocol a single command (read multiple holding registers) could be sent every 10s which would cause the sensor device to report values of both registers in a single response.

I am assuming the answer is 'No, Loxone always reads/writes registers or coils individually'. If that is the case, and has been confirmed, it would be a nice thing to note on the relevant document here.

Side note: Why do I think this is important? Well, I have built a Modbus environmental sensor which has different hardware modules to measure various environmental values. If one of these hardware modules fails I want to respond to a Modbus read register request with an error because the value isn't available due to hardware failure.

If Loxone reads multiple registers in a contiguous range, then to report an error on one register would actually be done by responding to the entire 'read multiple registers' request with an error. This would cause Loxone to not be able to read values from other sensor hardware that is still working.

If Loxone only reads single registers then the error would only be reported for the specific register that was unavailable (corresponding to the failed hardware sensor) and the rest would still work.

sarnau, 2022-08-03:
That is a complicated question. It seems Loxone is very transparent when it comes to Modbus, it just sends out the requests on a defined polling interval and forwards the responses back to the server. Modbus codes 1-4 (Read), 5-6, 15-16 (Write) and 7 (Read Exception Status) are supported.

I would read individual sensors with individual requests (single or multi, depending on the value range), otherwise I see no chance in detecting failures.

raintonr, 2022-08-03:
> I would read individual sensors with individual requests (single or multi, depending on the value range), otherwise I see no chance in detecting failures.

Yes, I would have thought that too, and yes, following a bit of testing, it is what Loxone seems to do.

When I connect my sensor to a Linux box and query using a `modpoll` read of multiple registers I can see the RX/TX lights on my senor's interface flash just one time each polling cycle. Ie. one request is received and one response is sent, and `modpoll` correctly shows the output from the multiple registers returned back.

I next connected my sensor to our Loxone Modbus extension and configured the system to read several registers, all at the same polling frequency. Two of these registers were contiguous addresses, and one of those two contiguous registers was intentionally 'broken' (the hardware sensor itself was disconnected). I know from testing with `modpoll` if one requests a range that includes the 'broken' register the entire request will fail and nothing will be returned.

In this scenario Loxone correctly read all the values aside from the 'broken' one and correctly reported that value was unavailable. Moreover, unlike the `modpoll` experience, I see the RX/TX lights flashing multiple times each polling cycle.

All this points to the fact that, no, Loxone never reads more than 1 register at a time and I would assume it's the same for coils, writes, etc. Could you please note this in the documentation as it's handy to know.